taligentx/dscKeybusInterface #289: Missing module interface data when using library dsckeybusinterface

Dilbert66, 2022-04-25:
Hi Nikhil, I was updating my library version to your latest develop branch and noticed that it broke a few things with my code.  This was related to the splitting off of the module data  read functionality out of the dscKeybusInterface.cpp code section to the  dscKeybusReader group only.    This is a concern since it is very useful to capture external module data such as wireless battery levels. The panel does not send those, but the modules do so it's really handy for that.   Could you not instead of just removing the code, just use a define to remove them if not needed but still keep the ability for the dsckeybusinterface to monitor both the panel and module data?

rburkholder, 2023-05-03:
Hi @Dilbert66, what are you doing in the meantime?  Do you include the dscKeybusReader group code?

Dilbert66, 2023-05-03:
I use my own implentation of this library for my own projects as I had a few specific requirements not supported here.