taligentx/dscKeybusInterface #294: question

george1616, 2022-06-14:
Hi,you could add the arming of the system with user code 01, 02 ......... 40  and disarming as well? And I would have another request if it is possible to transmit the periodic test is very useful. thank you and good luck

Alviunta, 2022-06-16:
Hi @george1616, if you want to track these events, I recommend that you investigate the protocol status notification system. Many of the states that you request are decoded in this library, I recommend that you look at the dscKeybusPrintData.cpp file, there you will find all the information you need. As a starting point, investigate the panel commands 0xA5, 0xCE, and 0xEB and the functions they connect to.
It is not a complicated task to do what you mention once you understand the basic operation.

Alviunta, 2022-06-24:
Hi @george1616, if you want to track the AC loss event, **you can use dsc.powerchange and dsc.powertrouble** just like you did in the example.
(I recommend that you look at the example files status.ino where you can find all the "variables" that the library tracks for you)

On the other hand, it may happen that some of the events that you want to track are not already tracked by the library, in that case you should do what I mentioned before. I leave you here an example of how to track the loss of AC:
`
...
if (dsc.loop()) // Returns true if valid panel data is available
{
switch (dsc.panelData[0])
        {
        // PANEL STATUS
        case 0xA5:
            // Partition: dsc.panelData[3] >> 6
            // Status set selector: byte 5 bits 0-1 (dsc.panelData[5] & 0x03)
            //  Status set 0 1 2 3
            //  Status byte 6 (dsc.panelData[6])
            setSelector = dsc.panelData[5] & 0x03;
            switch (setSelector)
            {
            case 0x00:
                switch (dsc.panelData[statusByte])
                {
                case 0xE8: // Panel AC power trouble
                     SOME CODE
                     break;
               case 0xF0: // Panel AC power restored
                     SOME CODE
                     break;
...
`
Short explanation: dsc.panelData[] is an array containing the bytes read directly from the keybus. With the information decoded (it is available to you in the file dscKeybusPrintData.cpp) and using the status messages, which are headed with 0xA0, 0xCE and/or 0xEB (hence the switch dsc.panelData[0]:) you can get panel status information. I strongly recommend you look at the file I mentioned, the information is extremely complete and there are very few unknown events.

If you want you can combine both ways to obtain the panel status.
If you want to understand a little more about how dsc.panelData[] is used you can look at the unlocker.ino example.
As a recommendation, only use dsc.panelData[] if you need to, since you'll have to spend some time understanding how it works and how to accomplish what you need to do.

I hope you find it useful. Sorry but I don't speak English well either.