taligentx/dscKeybusInterface #295: [Contribution] New Status on Set 0x16

Alviunta, 2022-06-16:
Hello.
I think I have figured out an event status not yet tracked.
When you program auto-disarm, regardless of the partition, an event status is generated with the following characteristics:
 - panelData[0] : 0xEB
 - use corresponding partition number
 - subset 0x16
 - subcommand 0x83

Here are some test logs I ran:
SELF DISARM
[0xEB (0x16) 0x83] 2022.06.16 12:45 | partition 2 | Unknown data - ERROR
[0xEB (0x16) 0x83] 2022.06.16 12:50 | partition 3 | Unknown data - ERROR
[0xEB (0x16) 0x83] 2022.06.16 12:55 | partition 4 | Unknown data - ERROR
[0xEB (0x16) 0x83] 2022.06.16 13:00 | partition 5 | Unknown data - ERROR
[0xEB (0x16) 0x83] 2022.06.16 13:05 | partition 6 | Unknown data - ERROR
[0xEB (0x16) 0x83] 2022.06.16 13:10 | partition 7 | Unknown data - ERROR
[0xEB (0x16) 0x83] 2022.06.16 13:15 | partition 8 | Unknown data - ERROR

It was tested on a PC1864 v 4.24 panel, for which you need to enter programming (*8) in sections 681 - 688.
It would be nice to have confirmation in other panels.
Greetings.

Alviunta, 2022-06-24:
Hi, I have a PC5200 module and its information was not decoded.
The panel is PC1864, it has capacity for 4 of these modules, so configure it in the 4 possible positions and try to obtain the supervision, battery, AC and tamper faults, most of them satisfactorily. I only had problems with slot 4 since I couldn't find how it is supervised or its tamper. (I suspect there are more bytes after this library can track, but I can't be sure)

Here is what I was able to discover:

pc 5200
SLOT 1

-Supervisory
  569.60: 00010001 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x11] Module supervision query
  569.60: 11111111 1 11111111 11111100 00000000 00001111 11111111 11111111 11001111 [Module/0x11] Keypad slots: 8 | Zone expander: 1 2 3 4 5 6 
  
-- Discon
  689.74: 00010001 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x11] Module supervision query
  689.74: 11111111 1 11111111 11111100 00000000 00001111 11111111 11111111 11111111 [Module/0x11] Keypad slots: 8 | Zone expander: 1 2 3 4 5 6 
  689.82: 10100101 0 00100010 00011010 11101100 01111111 00011111 00110011 10011110 [0xA5] 2022.06.23 12:31 | Unknown data
  689.88: 11001110 0 00100000 01111111 00011111 00000000 00000000 10001100 [0xCE] Unknown data
  690.09: 11101011 0 00000000 00100010 00011010 11101100 01111100 00000011 00011111 00110011 11100100 [0xEB] 2022.06.23 12:31 | Unknown data
  
-- Con
  809.74: 00010001 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x11] Module supervision query
  809.74: 11111111 1 11111111 11111100 00000000 00001111 11111111 11111111 11001111 [Module/0x11] Keypad slots: 8 | Zone expander: 1 2 3 4 5 6 
  809.83: 10100101 0 00100010 00011010 11101100 10000111 00011011 00110011 10100010 [0xA5] 2022.06.23 12:33 | Unknown data
  809.90: 11001110 0 00100000 10000111 00011011 00000000 00000000 10010000 [0xCE] Unknown data
  810.19: 11101011 0 00000000 00100010 00011010 11101100 10000100 00000011 00011011 00110011 11101000 [0xEB] 2022.06.23 12:33 | Unknown data
  
! pc 5200 supervisory slot 1 in [Module/0x11] byte 8 - bits 4,5

[Status set: (0x03)   Status: 0x1F] - | Unknown data   // Module suppervision loss - slot 1
[Status set: (0x03)   Status: 0x1B] - | Unknown data   // Module suppervision restore - slot 1

-Tamper

-- Activation
 1566.99: 11111111 1 11111111 11111111 11111110 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Module tamper notification 
 1567.17: 01001100 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x4C] Module tamper query
 1567.17: 11111111 1 11111111 11111111 11111111 11110000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 11110011 11111111 [Module/0x4C] Keypad tamper: Slot 8 
 1567.23: 10000111 0 00000000 00000010 10001001 [0x87] PGM outputs enabled: 2 
 1567.31: 10100101 0 00100010 00011010 11101100 10110111 01010111 01000101 00100000 [0xA5] 2022.06.23 12:45 | Unknown data
 1567.60: 11101011 0 00000000 00100010 00011010 11101100 10110100 00000011 01010111 01000101 01100110 [0xEB] 2022.06.23 12:45 | Unknown data

-- Restore
 1267.18: 01001100 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x4C] Module tamper query
 1267.18: 11111111 1 11111111 11111111 11111111 11110000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 11111100 11111111 [Module/0x4C] Keypad tamper: Slot 8 
 1267.23: 10000111 0 00000000 00000010 10001001 [0x87] PGM outputs enabled: 2 
 1267.31: 10100101 0 00100010 00011010 11101100 10100011 01010011 01000101 00001000 [0xA5] 2022.06.23 12:40 | Unknown data
 1267.60: 11101011 0 00000000 00100010 00011010 11101100 10100000 00000011 01010011 01000101 01001110 [0xEB] 2022.06.23 12:40 | Unknown data

! pc 5200 tamper slot 1 in [Module/0x4C] byte 13 - bits 0,1,2,3 -> tamper act 0011 - tamper restore 1100
? keypad tamper slot 8 ? keyboard without tamper capabilities -> 0000

[Status set: (0x03)  Status: 0x57] - | Unknown data   // tamper  - slot 1
[Status set: (0x03)  Status: 0x53] - | Unknown data   // tamper restore - slot 1

-AC

--AC loss
 2342.42: 11111111 1 11111111 11111111 11111111 11011111 11111111 11111111 11111111 11111111 [Module/0x05] Module status notification 
 2342.53: 01011000 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x58] Module status query
 2342.53: 11111111 1 11111111 11111111 11111111 11111111 00111111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x58] Unknown data
 2355.28: 11001110 0 00000010 11100000 11100000 11100000 11100000 01010000 [0xCE] Unknown data [Byte 2/0x02] 
 2355.38: 11101011 0 00000000 00100010 00011010 11101100 11101000 00010100 10100100 10101010 01011101 [0xEB] 2022.06.23 12:58 | Unknown data
 
--AC restore
 2520.05: 11111111 1 11111111 11111111 11111111 11011111 11111111 11111111 11111111 11111111 [Module/0x05] Module status notification 
 2520.17: 01011000 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x58] Module status query
 2520.17: 11111111 1 11111111 11111111 11111111 11111111 11001111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x58] Unknown data
 2534.94: 11101011 0 00000000 00100010 00011010 11101101 00000100 00010100 10100000 10101010 01110110 [0xEB] 2022.06.23 13:01 | Unknown data

[... cut ...]
			[Status set: (0x14)  Status: 0xAB] - | Unknown data   // battery restored - slot 4
		- PC5200 battery fail, slot 1 to 4: 0xAC, 0xAD, 0xAE, 0xAF
			[Status set: (0x14)  Status: 0xAC] - | Unknown data   // battery fail - slot 1
			[Status set: (0x14)  Status: 0xAD] - | Unknown data   // battery fail - slot 2
			[Status set: (0x14)  Status: 0xAE] - | Unknown data   // battery fail - slot 3
			[Status set: (0x14)  Status: 0xAF] - | Unknown data   // battery fail - slot 4
--||--

I hope it is useful and the meaning of each event is understood. My writing in English is not good.

kricon, 2022-06-26:
Auto-Disarm feature exists only on PC1616/PC1832/PC1864 panels, I've verified it with PC1832 v4.24, works exactly as you described.

Thanks for contribution with tracking all messages from PC5200, I've included it in PR #283 so you can download it from my repo and test it out as I don't have PC5200 module to verify.

Alviunta, 2022-06-28:
Hi @kricon , It's good to know that you were able to verify the auto-arming. Thank you!
After seeing what was already decoded regarding the 0x58 command, I realized that there was an over current protection. So I did the tests to see how the PC5200 notified the failure.
The results I got are consistent with what was seen on the PC5204. You can find them in the previous comment, they are not really many.
I will gladly test the changes you added. Cheers!

Alviunta, 2022-06-29:
Hi, i also decode fault states in zones 33 to 64. Fault occurs when there is a short circuit, same as decoded in set 0x01 states 0x6C - 0xAB. I share here the logs with the information.
Sorry for not sharing everything at once, I mistakenly short-circuited my zone expander and it was luck of the draw that I saw the message.
I also tested PR #283 and it works perfectly.
*-*-*-*-*-*--*-*-*-*-*-*-*-*-*-*-*
Zone expander 4 - 7
Zones fault (zones 33 - 64) (Zone in short circuit)

-Zone expander 4 ( 33 - 40 )
-- Zone 33
  153.71: 11111111 1 11111111 01010100 01010101 11111111 11111111 00111111 11111111 [Module/0xE6.08] Zone expander: 4 | Zones changed: 33 open 
  153.76: 11100110 0 00001001 00000001 11110000 [0xE6.09] Zones 33-40 open: 33 
  153.92: 11101011 0 00000000 00100010 00011011 10101001 01111000 00010100 01100000 01110010 00101111 [0xEB] 2022.06.29 09:30 | Unknown data
  154.00: 00011011 0 10000000 00000011 10000001 00000001 10000001 00000001 10000001 00000001 [0x1B] Partition 5: Backlight - Zones open | Partition 6: Ready Backlight - Partition ready | Partition 7: Ready Backlight - Partition ready | Partition 8: Ready Backlight - Partition ready
  154.47: 11100110 0 00101100 00000010 00000000 00000000 00000000 00000000 00010100 [0xE6.2C] Partition 2 | Enabled zones 33-64: none
  154.55: 10110001 0 11111111 00000000 00000000 00000000 00001000 11111111 00000000 00000000 10110111 [0xB1] Enabled zones 1-32 | Partition 1: 1 2 3 4 5 6 7 8 | Partition 2: 4 9 10 11 12 13 14 15 16 
  154.63: 00011011 0 10010000 00000011 10010001 00000001 10010001 00000001 10010001 00000001 [0x1B] Partition 5: Trouble Backlight - Zones open | Partition 6: Ready Trouble Backlight - Partition ready | Partition 7: Ready Trouble Backlight - Partition ready | Partition 8: Ready Trouble Backlight - Partition ready
  154.72: 00000101 0 10010001 00000001 10010001 00000001 10010001 00000001 10010001 00000001 [0x05] Partition 1: Ready Trouble Backlight - Partition ready | Partition 2: Ready Trouble Backlight - Partition ready | Partition 3: Ready Trouble Backlight - Partition ready | Partition 4: Ready Trouble Backlight - Partition ready
  159.57: 11000011 0 00001000 11111111 11001010 [0xC3] TLM: available/disabled | Dialer call attempt
  164.37: 11111111 1 11111111 11111111 11111111 11111111 11111111 01111111 11111111 11111111 [Module/0x05] Zone expander notification: 4 
  164.66: 11100110 0 00001000 11111111 11111111 11111111 11111111 11111111 11111111 [0xE6.08] Zone expander query: 4
  164.66: 11111111 1 11111111 01010101 01010100 11111111 11111111 00111111 11111111 [Module/0xE6.08] Zone expander: 4 | Zones changed: 33 closed 
  164.73: 11100110 0 00001001 00000000 11101111 [0xE6.09] Zones 33-40 open: none 
  165.05: 11101011 0 00000000 00100010 00011011 10101001 01111000 00010100 01000000 01110010 00001111 [0xEB] 2022.06.29 09:30 | Unknown data
[Selector: (0x14) Status: 0x60] | Unknown data - ERROR // zone fault 33
[Selector: (0x14) Status: 0x40] | Unknown data - ERROR // zone fault restore 33

-- Zone 34
  313.67: 11111111 1 11111111 01010001 01010101 11111111 11111111 00001111 11111111 [Module/0xE6.08] Zone expander: 4 | Zones changed: 34 open 
  313.75: 11100110 0 00001001 00000010 11110001 [0xE6.09] Zones 33-40 open: 34 
  314.09: 11101011 0 00000000 00100010 00011011 10101001 10000100 00010100 01100001 01110010 00111100 [0xEB] 2022.06.29 09:33 | Unknown data
  314.17: 00011011 0 10000000 00000011 10000001 00000001 10000001 00000001 10000001 00000001 [0x1B] Partition 5: Backlight - Zones open | Partition 6: Ready Backlight - Partition ready | Partition 7: Ready Backlight - Partition ready | Partition 8: Ready Backlight - Partition ready
  314.49: 00011011 0 10010000 00000011 10010001 00000001 10010001 00000001 10010001 00000001 [0x1B] Partition 5: Trouble Backlight - Zones open | Partition 6: Ready Trouble Backlight - Partition ready | Partition 7: Ready Trouble Backlight - Partition ready | Partition 8: Ready Trouble Backlight - Partition ready
  314.57: 00000101 0 10010001 00000001 10010001 00000001 10010001 00000001 10010001 00000001 [0x05] Partition 1: Ready Trouble Backlight - Partition ready | Partition 2: Ready Trouble Backlight - Partition ready | Partition 3: Ready Trouble Backlight - Partition ready | Partition 4: Ready Trouble Backlight - Partition ready
  314.58: 11111111 1 11111111 11111111 11111111 11111111 11111111 01111111 11111111 11111111 [Module/0x05] Zone expander notification: 4 
  314.84: 11100110 0 00001000 11111111 11111111 11111111 11111111 11111111 11111111 [0xE6.08] Zone expander query: 4
  314.84: 11111111 1 11111111 01010101 01010001 11111111 11111111 00001111 11111111 [Module/0xE6.08] Zone expander: 4 | Zones changed: 34 closed 
  314.91: 11100110 0 00001001 00000000 11101111 [0xE6.09] Zones 33-40 open: none 
  315.23: 11101011 0 00000000 00100010 00011011 10101001 10000100 00010100 01000001 01110010 00011100 [0xEB] 2022.06.29 09:33 | Unknown data
[Selector: (0x14) Status: 0x61] | Unknown data - ERROR // zone fault 34
[Selector: (0x14) Status: 0x41] | Unknown data - ERROR // zone fault restore 34

-- Zone 35
 1266.22: 11111111 1 11111111 01000101 01010101 11111111 11111111 00111111 11111111 [Module/0xE6.08] Zone expander: 4 | Zones changed: 35 open 
 1266.29: 11100110 0 00001001 00000100 11110011 [0xE6.09] Zones 33-40 open: 35 
 1266.66: 11101011 0 00000000 00100010 00011011 10101001 11000100 00010100 01100010 01110010 01111101 [0xEB] 2022.06.29 09:49 | Unknown data
[... cut ...]
[Selector: (0x14) Status: 0x7F] | Unknown data - ERROR // zone fault 64
[Selector: (0x14) Status: 0x5F] | Unknown data - ERROR // zone fault restore 64

Zone expander 7:

	Selector: 0x14 
		- Status : 0x58 - 0x5F : Zone fault restore 57 - 64
		- Status : 0x78 - 0x7F : Zone fault 57 - 64
---------------------------------------------------------------
Zone expander 4 - 7:
	Selector: 0x14
		- Status: 0x40 - 0x5F // Zone fault restore 33 - 64
		- Status: 0x60 - 0x7F // Zone fault  33 - 64
---------------------------------------------------------------