taligentx/dscKeybusInterface #303: PC3000 Testing

b0bcarlson, 2022-08-21:
I am currently setting up Arduino Uno to a PC3000.
Ran the KeybusReader without issue using the instructions for classic wiring. Switching to the MQTT I am having some issues but I believe they are network related (struggling to connect to the MQTT server when it worked before wiring it). I will update when I have updates!

b0bcarlson, 2022-08-21:
I was able to get it connected to MQTT and it shows in HA as disarmed, however I cannot seem to get anything else to work. No zones update in HA and attempting to arm from HA does not do anything. Guessing this is a MQTT config issue so I will continue to investigate

b0bcarlson, 2022-08-23:
I realized the unlocker is not compatible with the classic series, so I modified it to brute force and was able to get the installer code. Unfortunately I am having connection issues again, but I'm thinking things are still looking pretty good.

b0bcarlson, 2022-08-26:
After continuing to mess around with this I have come up with a pattern:
- enabling PC16 output breaks networking (DHCP failures, MQTT connection issues)
- disabling PC16 output makes things almost work - pressing "Arm" triggers a single key press (instead of 4), but more importantly sensors don't seem to communicate correctly (shows most as unknown)
- disabling PC16 output and commenting out `#define dscClassicSeries` behaves similarly to the above except that the arm buttons do not trigger any key presses.
I'll keep this ticket open for now in case anyone has any ideas. However, this is not my house and I will be moving in a week so I will not be able to test after that.

sfossen, 2022-08-27:
Have you tried [a modified version](https://github.com/taligentx/dscKeybusInterface/pull/296)?

b0bcarlson, 2022-08-27:
Yes, actually I saw that yesterday and started testing it with much better luck. Still not perfect though; I moved the PC16 from 4 to 12 and the network issues stopped but when I use MQTT it seems to cause interference (breadboard probably). I haven't had much time to get further but I was going to try a few other modifications and see where they get me.

sfossen, 2022-08-27:
Which version of the circuit did you use? The version I pointed you to was developed on a breadboarded circuit with optocouplers.

kricon, 2022-09-04:
@b0bcarlson sorry that im late here,
- For PC3000 you need to use sfossen's PC3000 library, because PC3000 uses 3 bytes of keybus data (where PC1550/PC2550 have just two bytes and PC560 just one). 
- You also **need** to have `#define dscClassicSeries` **uncommented** for classic series to work. Otherwise, you will get junk and false data and things just wont work as whole classic/powerseries structure is so much different.
- I've had better luck with just twisting wires to resistors and transistor for quick bench-testing than using cheapo bad quality breadboards before soldering onto protoboard.
- PC16 out **must** be enabled (section 19, option 4) aswell as PGM out set to Strobe output (section 28, second digit 8).
- For unlocker sketch to works with classic series, you must use sfossen's library, uncommented define and have system in ready state. Unlocker example works fine for me on PC1550 and I belive that sfossen also tested it with his PC3000 panel.
- Try `KeybusReader` sketch (from sfossen's PC3000 library) through USB serial and post logs here if things doenst work as it should.
- On Arduino IDE, if you have both dscKeybusInterface and dscKeybusInterface_PC3000 libraries in Documents/Arduino folder, PC3000 example will use source data from regular dscKeybusInterface. Delete or temorary move dscKeybusInterface folder from Documents/Arduino/libraries for PC3000 library to work.