taligentx/dscKeybusInterface #318: Keybus Error / Address?

whiteas2, 2023-05-20:
Hello,

I have a PC1864. I am getting a keybus error once I connect the ESP to the keybus. The keypads show an  error. I can silence the error but can do nothing else on the keypad. Supervision is off. I suspect that the problem may be the address of the unit is clashing with a keypad address. Is some one able to tell me what address is used; and also where I can change it (if possible)?

The ESP was tested on a different alarm panel and it worked so I know the device is ok.

Dilbert66, 2023-05-21:
This all points to a bad wiring connection from your esp to the panel.  There is no address to configure.

whiteas2, 2023-05-22:
> This all points to a bad wiring connection from your esp to the panel. There is no address to configure.

Hi. Thank you for the response. I will check all and try again.

l33d, 2023-05-22:
Yes this is bad connection between esp and keybus. Check all connection.