taligentx/dscKeybusInterface #324: DSC PC1550 Clasic Series 6 digit code compile ok but keep rebooting

pansament, 2023-08-19:
### Discussed in https://github.com/taligentx/dscKeybusInterface/discussions/323

<div type='discussions-op-text'>

<sup>Originally posted by **pansament** August 15, 2023</sup>
I have managed to determinate that whet i set in sketch "accesCode" my 6 digit arm/disarm code in serial monitor it keeps rebooting.

If code not set everything is ok.

Do i need to configure alarm to work with 4 digit code? 
Or what can i modify in sketch to work with 6 digit code?

This is error code from serial monitor:

> `⸮⸮n⸮r⸮⸮n|⸮l⸮l`bbrl⸮nb⸮nl`⸮rl⸮l⸮⸮
> 
> Static IP Configured
> WiFi........connected: 25.25.25.151
> MQTT....connected: 25.25.25.50
> DSC Keybus Interface is online.
> 
> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
> 
> Exception (4):
> epc1=0x400005cb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
> 
> >>>stack>>>
> 
> ctx: cont
> sp: 3ffffe00 end: 3fffffd0 offset: 0150
> 3fffff50:  3fffdad0 3fffff80 40205d66 3fffff80  
> 3fffff60:  3fffdad0 00000000 3ffeedc0 40204b69  
> 3fffff70:  3fffdad0 00000000 3ffeedc0 40201d5f  
> 3fffff80:  3ffe9010 fffffffc 00000080 40208848  
> 3fffff90:  40204e8c 3ffe900e 3ffef048 3ffef0f0  
> 3fffffa0:  3fffdad0 0000001f 3ffef048 3ffef0f0  
> 3fffffb0:  3fffdad0 00000000 3ffef0c4 40205efc  
> 3fffffc0:  feefeffe feefeffe 3fffdab0 401014e9  
> <<<stack<<<
> 
> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3424, room 16 
> tail 0
> chksum 0x2e
> load 0x3fff20b8, len 40, room 8 
> tail 0
> chksum 0x2b
> csum 0x2b
> v00052840
> ~ld
> `</div>

kricon, 2024-03-08:
Hi @pansament, sorry for late answer.
I took a look at code and you could try manually editing line `712` in `dscKeybusInterface/src/dscClassic.cpp`:
`if (strlen(accessCodeStay) < 4) {` Try increasing it from 4 to 6.

As I don't have any 6-digit ClassicSeries panels available to confirm I'm hoping to hear back from you, so we can implement AccessCode string lenght checking and automatically setting 4 or 6 digit Access Code for ClassicSeries.

pansament, 2024-03-08:
> Hi @pansament, sorry for late answer. I took a look at code and you could try manually editing line `712` in `dscKeybusInterface/src/dscClassic.cpp`: `if (strlen(accessCodeStay) < 4) {` Try [...]

Hi @kricon, Thanks for the answer!
I`ve tried now to increase to 6 and test it and I`m getting same behavior/error.

I`m searching to buy a PC1616 board to replace what I have now if I cannot work with this board.