taligentx/dscKeybusInterface #328: Debugging disarm not working overnight

bobfang, 2023-09-28:
Hello! Great work with the keybus! I would like to receive some pointers on how to debug my setup.

I'm having an issue where my homebridge MQTT setup is unable to disarm the DSC PC1832 after being armed overnight.  I only have 1 partition.

What works:
- Arming system
- Receiving updates via MQTT for all zones
- Disarming if armed in the last 10-20 minutes

Does not work:
- Disarm after being in armed stay overnight (access code is saved in the sketch). I have to use the panel manually to disarm.

I'm using one of the gerber files fabricated by pcbdirect so the wiring within the ESP8266 should be within spec.  Power is supplied using a buck converter to step 12v to 3.3v on the esp8266.

Based on the code it just looks like the disarm portion is simply achieved by sending the access code down the keybus

Thank you!

armandvanderwalt, 2023-12-31:
I have the exact same issue, disarming a normal away arm with no issues, but after about 30 mins I cannot disarm stay and have to use the keypad to manually disarm.

bobfang, 2024-01-10:
@armandvanderwalt what kind of transistor do you have on your board? i'm using a 2N3904

feinadam, 2024-03-30:
Any solution? I have exactly the same issue.

luis1424, 2025-06-29:
Hello, I'm experiencing the same problem. I've found that this happens if the MQTT connection is lost and reconnected. After that no arming nor disarming is possible. I tried resetStatus() but does noy work. Has someone found a solution?

bobfang, 2025-10-20:
Recently, I've switched to using the Home Assistant MQTT version of the sketch provided in the examples and this issue went away completely. Same hardware was used for both homebridge and home assistant.

One could technically use the home assistant sketch and then change their configuration in homebridge MQTT things to read the new topics to get around this issue.

Hope this helps someone!