taligentx/dscKeybusInterface #331: Lastest version outgrew UNO??

joeinaustin, 2023-12-13:
Just getting started with this but immediately ran into an issue on verify in Arduino IDE.

"Sketch uses 32322 bytes (100%) of program storage space. Maximum is 32256 bytes.
Global variables use 1017 bytes (49%) of dynamic memory, leaving 1031 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board"

It looks like it is now too big for many of the Arduino boards.  It compiles on the Mega fine.  

Am I missing something here?  I see no references to this in the discussions.

joeinaustin, 2023-12-21:
I need to clarify this.  The issue is specific to the KeyBusReader example.  It just happened to be the first example I tried.  Some of the other examples I tried did not have this issue.  At a minimum, this issue should be included in that examples code notes.  I have not figured out how many of the examples have this issue.

kricon, 2024-03-08:
Issue is fixed in latest [develop branch](https://github.com/taligentx/dscKeybusInterface/tree/develop)
You need to download it and manually place into \Documents\Arduino\libraries folder.

taligentx, 2024-03-18:
Hi @joeinaustin - as @kricon noted, this is fixed in `develop` so the next release (yep, there will be one) will take care of it for installs through the Arduino and PlatformIO library registries.  Pretty obvious bug that I should have caught in testing.