taligentx/dscKeybusInterface #338: PC3000 - Zones 9-16 not updating

agl1tch3dpir8, 2024-02-18:
Zones 1-8 work fine... I also need help programming a reboot every couple of days. After 1-2 days zones 1-8 also stop updating. Any help is greatly appreciated.

kricon, 2024-03-08:
Hi @agl1tch3dpir8 sorry for late answer.

PC3000 panel uses 3 bytes, one byte more than other panels, which are used for zone 9-16 reporting.
**There is PR #296 for PC3000 zones 9-16 support. You can download it [here](https://github.com/sfossen/dscKeybusInterface/tree/DSC_PC3000). It should fix your problem of zones 9-16 not updating.**

If I remember correctly, that PR works fine also on other ClassicSeries (PC1550/PC2550) so hopefully @taligentx will merge it soon and publish a new release as recently there was many questions about PC3000 support and some Arduino sketch size issues in master branch...

sfossen, 2025-01-03:
@taligentx I've been running the code on my own pc-3000 successfully, and would also like some external confirmation. :)