taligentx/dscKeybusInterface #344: Issues compiling ESP32 examples

AlFreddyTo, 2024-06-09:
Good day.  

I am able to compile in ARduino IDE 2.3.2 the Telegram example for ESP8266, but I am getting a lot of timer errors when compiling the Telegram example for ESP32. See below. Any solution for this ? Tks

`c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.cpp: In member function 'void dscClassicKeypadInterface::begin(Stream&)':
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.cpp:65:22: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
   65 |   timer1 = timerBegin(1, 80, true);
      |            ~~~~~~~~~~^~~~~~~~~~~~~
In file included from C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal.h:84,
                 from C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/Arduino.h:36,
                 from c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.h:23,
                 from c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.cpp:20:
C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.cpp:67:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
   67 |   timerAttachInterrupt(timer1, &dscClockInterrupt, true);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.cpp:68:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
   68 |   timerAlarmWrite(timer1, 1000, true);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassicKeypad.cpp:69:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
   69 |   timerAlarmEnable(timer1);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp: In member function 'void dscClassicInterface::begin(Stream&)':
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp:67:22: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
   67 |   timer1 = timerBegin(1, 80, true);
      |            ~~~~~~~~~~^~~~~~~~~~~~~
In file included from C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal.h:84,
                 from C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/Arduino.h:36,
                 from c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.h:23,
                 from c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp:20:
C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp:69:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
   69 |   timerAttachInterrupt(timer1, &dscDataInterrupt, true);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp:70:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
   70 |   timerAlarmWrite(timer1, 250, true);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp:71:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
   71 |   timerAlarmEnable(timer1);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp: In member function 'void dscClassicInterface::stop()':
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscClassic.cpp:92:3: error: 'timerAlarmDisable' was not declared in this scope
   92 |   timerAlarmDisable(timer1);
      |   ^~~~~~~~~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeypad.cpp: In member function 'void dscKeypadInterface::begin(Stream&)':
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeypad.cpp:65:22: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
   65 |   timer1 = timerBegin(1, 80, true);
      |            ~~~~~~~~~~^~~~~~~~~~~~~
In file included from C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal.h:84,
                 from C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/Arduino.h:36,
                 from c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeypad.h:23,
                 from c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeypad.cpp:20:
C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeypad.cpp:67:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
[... 23 lines cut ...]
      |             ^~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeybusInterface.cpp:67:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
   67 |   timerAttachInterrupt(timer1, &dscDataInterrupt, true);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeybusInterface.cpp:68:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
   68 |   timerAlarmWrite(timer1, 250, true);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeybusInterface.cpp:69:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
   69 |   timerAlarmEnable(timer1);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeybusInterface.cpp: In member function 'void dscKeybusInterface::stop()':
c:\Users\Freddy\Documents\Arduino\libraries\DSC_Keybus_Interface\src\dscKeybusInterface.cpp:90:3: error: 'timerAlarmDisable' was not declared in this scope
   90 |   timerAlarmDisable(timer1);
      |   ^~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\Freddy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\WiFi
  Not used: C:\Users\Freddy\Documents\Arduino\libraries\WiFi
exit status 1

Compilation error: exit status 1`

kpostekk, 2024-06-25:
I got same error after upgrading board manager to 3.0.x.
Downgrading board manager to 2.0.17 allows program to compile.

kpostekk, 2024-06-25:
[image]

AlFreddyTo, 2024-06-26:
Thanks very much for the info !