taligentx/dscKeybusInterface #349: PC1500 Classic Series armed/disarm

rejavarti, 2024-07-15:
Good morning, 

An issue that I am having with the setup is that it will show disarmed and also show pending, but will never change from pending to armed (either home or away) did some sleuthing with the status and keybus examples and found that when the ready light goes out to let know that the system is armed, the status example doesn’t switch, but the keybus shows that it sees the ready light go out. 

What it is looking like is that the status is not picking up that the ready light went out and not changing the status from pending to armed. 

Hope that this can help with diagnosing the issue.

zayan548, 2024-07-15:
Code for house security in ai

rejavarti, 2024-07-15:
> Code for house security in ai

What do you mean by that? Thanks

zayan548, 2024-07-15:
I wanna complete code for house security

justbendev, 2024-08-23:
Sorry I'm having difficulties understanding you.
From what i have understood you are having difficulties using the light status to know if the system is Armed / Ready.

**I would suggest using one of theses data point :**
```c++
dsc->armed[partition]
dsc->armedAway[partition]
dsc->armedStay[partition]
dsc->exitDelay[partition]
dsc->ready[partition]
```
_note that armed(...) will only become true after the ExitDelay_

rejavarti, 2024-08-23:
Thanks for the reply, in home assistant, it's just flashing pending the whole time it is either armed home, or away. It is not recognizing the exit delay and switching to the proper armed mode. 

This is my MQTT: 

# https://www.home-assistant.io/integrations/alarm_control_panel.mqtt/
  alarm_control_panel:
    - name: "DSC 1500 Alarm Panel"
      unique_id: dscPartition1
      state_topic: "dsc/Get/Partition1"
      code: REMOTE_CODE
      command_template: >
        {% if action == "1D" %}
          1!{{ code }}
        {% else %}
          {{ action }}
        {% endif %}
      payload_disarm: "1D"
      payload_arm_home: "1S"
      payload_arm_away: "1A"
      payload_arm_night: "1N"
      availability_topic: "dsc/Status"
      command_topic: "dsc/Set"
      supported_features:
        - arm_home
        - arm_away
        - arm_night
      code_arm_required: true
      code_disarm_required: true
      code_trigger_required: true

  # https://www.home-assistant.io/integrations/sensor.mqtt/
  sensor:
    - name: "Security Partition 1"
      unique_id: dscPartitionMessage1
      state_topic: "dsc/Get/Partition1/Message"
      icon: "mdi:shield"
      availability_topic: "dsc/Status"

  # https://www.home-assistant.io/integrations/binary_sensor.mqtt/
  binary_sensor:
    - name: "Security Trouble"
      unique_id: dscTrouble
      state_topic: "dsc/Get/Trouble"
      device_class: "problem"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "Downstairs Smoke Alarm"
      unique_id: dscFire1
      state_topic: "dsc/Get/Fire1"
      device_class: "smoke"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "Back Door"
      unique_id: dscZone1
      state_topic: "dsc/Get/Zone1"
      device_class: "door"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "Front Door"
      unique_id: dscZone2
      state_topic: "dsc/Get/Zone2"
      device_class: "door"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "Upper Patio Door"
      unique_id: dscZone3
      state_topic: "dsc/Get/Zone3"
      device_class: "door"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "Downstairs Motion"
      unique_id: dscZone4
      state_topic: "dsc/Get/Zone4"
      device_class: "motion"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "Kitchen Glass Break"
      unique_id: dscZone5
      state_topic: "dsc/Get/Zone5"
      device_class: "tamper"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "PGM 1"
      unique_id: dscPGM1
      state_topic: "dsc/Get/PGM1"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

    - name: "PGM 8"
      unique_id: dscPGM8
      state_topic: "dsc/Get/PGM8"
      payload_on: "1"
      payload_off: "0"
      availability_topic: "dsc/Status"

  # https://www.home-assistant.io/integrations/button.mqtt/
  button:
    - name: "Fire Alarm"
      unique_id: dscFire
      command_topic: "dsc/Set"
      payload_press: "f"
      icon: "mdi:fire"
      availability_topic: "dsc/Status"

    - name: "Aux Alarm"
      unique_id: dscAux
      command_topic: "dsc/Set"
      payload_press: "a"
      icon: "mdi:hospital-box"
      availability_topic: "dsc/Status"

    - name: "Panic Alarm"
      unique_id: dscPanic
      command_topic: "dsc/Set"
      payload_press: "p"
      icon: "mdi:police-badge"
      availability_topic: "dsc/Status"

justbendev, 2024-08-24:
Hi :wave: , Im not familiar with home assistant unfortunately but im going to try my best to help you figure it out.

A quick glance at your config files and something pretty obvious come to my mind, are you sure you are arming partition 1 ?

In your config files you set it up to only monitor partition 1 but ExitDelay is different for each partition its not shared variable so if your keypad is arming partition 2 you won't receive it with current config.
_DSC Panels supported by this library can have up to 8 partitions_

Also for debugging purposes i would suggest installing a software like MQTTX to read the MQTT Messages being sent to understand exactly the problem. Subscribe to dsc/# to catch everything coming out of the Microcontroller.

This part of the code in the HomeAssistant MQTT.ino is responsible for sending the "pending" topic with MQTT.
So the code definitely has the feature.
```c++
// Publishes exit delay status
      if (dsc.exitDelayChanged[partition]) {
        dsc.exitDelayChanged[partition] = false;  // Resets the exit delay status flag
        char publishTopic[strlen(mqttPartitionTopic) + 2];
        appendPartition(mqttPartitionTopic, partition, publishTopic);  // Appends the mqttPartitionTopic with the partition number

        if (dsc.exitDelay[partition]) mqtt.publish(publishTopic, "pending", true);  // Publish as a retained message
        else if (!dsc.exitDelay[partition] && !dsc.armed[partition]) mqtt.publish(publishTopic, "disarmed", true);
      }
  ```
  
  The publishTopic is defined as follows : _Basically it would be = "dsc/Get/Partition1/Message"_
  ```c++
 // MQTT topics - match to Home Assistant's configuration.yaml
const char* mqttClientName = "dscKeybusInterface";
const char* mqttPartitionTopic = "dsc/Get/Partition";  // Sends armed and alarm status per partition: dsc/Get/Partition1 ... dsc/Get/Partition8
const char* mqttPartitionMessageSuffix = "/Message";   // Sends partition status messages: dsc/Get/Partition1/Message ... dsc/Get/Partition8/Message
const char* mqttZoneTopic = "dsc/Get/Zone";            // Sends zone status per zone: dsc/Get/Zone1 ... dsc/Get/Zone64
const char* mqttFireTopic = "dsc/Get/Fire";            // Sends fire status per partition: dsc/Get/Fire1 ... dsc/Get/Fire8
const char* mqttPgmTopic = "dsc/Get/PGM";              // Sends PGM status per PGM: dsc/Get/PGM1 ... dsc/Get/PGM14
const char* mqttTroubleTopic = "dsc/Get/Trouble";      // Sends trouble status
const char* mqttStatusTopic = "dsc/Status";            // Sends online/offline status
const char* mqttBirthMessage = "online";
const char* mqttLwtMessage = "offline";
const char* mqttSubscribeTopic = "dsc/Set";            // Receives messages to write to the panel
 ...
 publishMessage(mqttPartitionTopic, partition); // Basically mqttPartitionTopic = sourceTopic
  ...
  itoa(partition + 1, partitionNumber, 10);
  strcpy(publishTopic, sourceTopic);
  strcat(publishTopic, partitionNumber);
  strcat(publishTopic, mqttPartitionMessageSuffix);
  ```

rejavarti, 2024-08-24:
Ok, just looked at the MQTT again using MQTT Explorer, and even in MQTT, it shows as pending, So it almost looks like it could be an issue with the microcontroller code.

[image]

 It is Partition 1 that I am using. and from what I am seeing, is that the Partition ready message in the MQTT is not changing from Ready. 

[image]

I'm just looking at the .ino file myself, and I am on the right partition (#1) for my system.