taligentx/dscKeybusInterface #351: Force data synchronization [PC5010] (Power832)

justbendev, 2024-08-21:
#### Hello :wave: 

I was wondering if there is any way to trigger a full panel sync ?

For example when the Panel is started before the microcontroller the OpenZones aren't accurate for a long time.

I tried the "L" _LCD keypad data request_ and "R" _Reset_ , both do not update the openZones.

This is especially problematic when the system is armed and disarmed with bypassed zones.

All zones are then marked not open when disarmed. (Basically openZones is empty/last state)

I was wondering if there is a way to query partition state ?

Thanks everyone :+1:

Dilbert66, 2024-08-21:
You can go into installer mode then exit. This will trigger the update.  For example, if your installer code is 1234 you can send:
*81234##

justbendev, 2024-08-22:
@Dilbert66 , thanks for the tip :+1: 

Will try that as soon as possible, if this works it's already way better than having to wait 5 to 10 mins for the panel to send a sync packet. 

Im doing a similar trick to retrieve bypassed zones (*11234#) and troubles (*21#7##)

Thanks alot :wave:

justbendev, 2024-08-23:
@Dilbert66 Solution works so im closing this :+1:

closed 2024-08-23 by justbendev