taligentx/dscKeybusInterface #355: DSC Unlocker 1.3 (esp32) - Stuck in dsc.loop()

jndrouin53, 2024-10-09:
First, I'm a new at this so I may have made some newbee mistakes somewhere.

I'm trying to find the installer code on a DSC PC1500 installed back in 1988 in the hope of integrating my old system into HA.

So the Unlocker 1.3 program appears to get stuck in the dsc.loop().  I presume it is not getting the proper (or any) dsc.status() from the board.  

Here's my setup.  I'm using a Freenove kit with an Espressif ESP32-Wroom-32E chip.  I've followed the wiring instructions as best as I could and soldered everything.  In my sketch I use the esp32 v2.0.17 board  (version 3.x will not compile).  In the preferences in the "Additional board managers URLs" I have specified "https://dl.espressif.com/dl/package_esp32_index.json" and "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json".  I've successfully compiled and loaded Unlocker 1.3 program in the ESP32 chip using either the ESP2-WROOM-DA Module and the ESP32-Dev Module.  I've connected Pin 18 to DSC Yellow, PIN19 to DSC Green and also PIN21 to DSC Green via a 2N3904 NPN transistor.   I should point out that I also have prepared PIN17 for eventual connection to PGM/AUX(+).  So PIN17 IS CONNECTED to a 10k ohm resistor that goes to a ground pin AND to a 33k ohm resistor that is connected to a 1k ohm resistor that goes nowhere.

My question:  is Unlocker 1.3 able to connect to the DSC Keybus on a PC1500 board (assuming that PGM is configured with PC16OUT disabled, the default)?  If so, what am I doing wrong?

kricon, 2024-10-11:
Hi @jndrouin53! For DSC Unlocker support on Classic-series, you need unlocker v1.4 which is available from the [develop](https://github.com/taligentx/dscKeybusInterface/tree/develop) branch.

And also to enable Classic series support you need to **un-comment line 104**, remove // from #define dscClassicSeries.

As for PC16 PGM out, im not quite sure if thats needed for unlocking purposes, I know PC16OUT gives out more information like system armed/trouble and without it it's just keypad status/zone lights and beeps which may be enough for unlocked sketch to know when its in programming mode.

jndrouin53, 2024-10-13:
Well...  no luck unfortunately.  Not being familiar with this, I thought I'd provide you with some info on my setup.  I'm using Arduino IDE 2.3.3 using board esp32 version 2.0.17 and libraries DSC Keybus Interface version 3.0.  

I added in the sketch the following line of code to see if any status was coming to the sketch from the 1500 board: "Serial.println(dsc.status[0])".  But nothing is printing out on the serial output, either because my syntax is flawed (I'm new at this) or because the dscKeybusInterface is not getting anything from the board.

To be on the safe side, I rechecked all my connections, from the pin on the ESP32 board to the tip of the wire that I screw on the board and the resistance I see corresponds to the resistors that I soldered so I assume that all my connections are good.

Here's the serial output after I reset the ESP32 board.
. - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - . - .
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

[  1011][W][esp32-hal-timer.c:226] timerAttachInterruptFlag(): EDGE timer interrupt is not supported! Setting to LEVEL...
DSC Keybus Interface....

kricon, 2024-10-13:
I'm not familar with Arduino IDE 2.x version, but if it's the same as 1.x you need to git clone the repository and then manually switch (git checkout develop) to develop branch, or download .zip from [develop branch here](https://github.com/taligentx/dscKeybusInterface/archive/refs/heads/develop.zip) and then extract it and replace contents in your Arduino libraries folder (tipically Documents/Arduino/libraries), overwritting the default ones.
Then from examples\esp32 folder open either Unlocker or KeybusReader example and **uncomment _#define dscClassicSeries_**.

I'm suggesting you to try KeybusReader at first to verify that wiring/esp32 communication with alarm panel is correct, and if it sees keybus data, follow keypad lights and you can send message input from terminal to panel (for example try arming/disarming with entering your access code in terminal and sending it), you can go ahead with Unlocker sketch. Note that Unlocker won't work if system is armed.

jndrouin53, 2024-10-13:
Just to clarify, after your initial reply suggesting that I use Unlocker from the "develop" branch, I went into the "develop" branch online, copy/pasted the Unlocker 1.4 source code into my sketch and uncommented the "#define dscClassicSeries" line as suggested.  So my 2nd attempt was done using the Unlocker 1.4 code.  BUT!!!

What I did NOT do is what you just suggested, ie. overwrite the whole "DSC_Keybus_Interface" library with the one from the "develop" branch which, I presume, contains changes in the .cpp and .h files.

I'll proceed to do that and I'll let you know the results.