taligentx/dscKeybusInterface #357: dsc.loop() in a timer event interrupt

cskzoli, 2024-10-14:
Hi,

Asking advice on following:
In my main loop I have some blocking procedures (can be several seconds) and would like to put dsc.loop() out from the main loop (to maintain its functionality).
My idea is to run dsc.loop() with a timer interrupt (eg. every 100ms).

What is the recommended interval for dsc.loop() to run?
Developing on PlatformIO/esp32 if that matter.

Thanks

Dilbert66, 2024-10-14:
Just run it as a task on the second core.
https://techtutorialsx.com/2017/05/09/esp32-running-code-on-a-specific-core/

cskzoli, 2024-10-14:
Hi Dilbert66
Good idea, but not sure that dsc.loop() function is designed in a thread safe operation (eg. dsc.loop put something into the dsc buffer, while main loop also accessing the same buffer...)
Thx

Dilbert66, 2024-10-14:
Might be an issue but since you are only reading the buffer with the main loop and not writing to it, it might work be fine. Yes, you are also resetting also some flags so that could be a cause of issue.   

Alternatively, you could instead put your blocking process in it's own task instead.